eh2k/squares-and-circles #123: Distorted right hand channel in resonator synth mode

hotstuffwonkyair, 2025-03-03:
Hello,
  Not sure if this could be hardware issue but judging from other modes I don’t think so,
 Using ( for instance ) the NES noise channel patched into C4 ( or any C channel ) as a stimulator source for the resonant mode, then having that mode output to A+B as stereo , with the input signal centered in the pan the A channel is clear and fine but the B channel is quiet and distorted..
 As I am not sure as to how much of the ‘stereo’ feature of Rings is included in this mode , it’s hard to gauge if MISO mono in stereo out should function this way? 
Prob I need to test with other sources and if resonator without audio input sounds the same.
Anybody else get this?

hotstuffwonkyair, 2025-03-03:
Ah, ok I think I’m getting it, the resonator does randomize the stereo but only on trigs , so unlike in Rings the audio input signal doesn’t quite stimulate that function.. so you can end up with just a left channel and a tiny bit in right channel until you throw a trigger at it..
 So it’s sort of a feature but also a slight inconvenience if you don’t want it bouncing around ..

eh2k, 2025-03-03:
@hotstuffwonkyair  - the rings/resonator engine is a bit special. It's actually a mixture of a synth and an effect. I don't know exactly how it is with a real rings - but here you always need a trig+cv input (like other synths) and optionally an audio input (like other fx engines) - the sound appears alternately on the left and right audio channel...