eh2k/squares-and-circles #124: Polyphony for the Rings machine

koswir, 2025-03-16:
Hey @eh2k 

The original Rings has the poly button.
"The first button selects the polyphony of the module: one, two, or four notes. Enabling four notes polyphony doesn't mean that four CV input jacks will magically appear on the module, but simply that four notes played in sequence will nicely overlap without cutting each other's tails. To play chords, you will need to "strum" the module by playing a rapid sequence of notes - something you might have already encountered with Braids' PLUK model. Note that the module might reduce the number of harmonics in the generated signals to cope with the higher polyphony."
It splits the 60 filters into 2x30 or 4x15 to do polyphony.

https://pichenettes.github.io/mutable-instruments-documentation/modules/rings/manual/

Any chance for implementing the poly version?

eh2k, 2025-03-17:
Hi coswir,

I took a closer look at the rings polyphony setting in vcv. In short, pressing the button switches polyphony to 1, 2, 4.

In squares-and-circles polythony is fixed to 3.
I did this back then to save memory and cpu time. But I can no longer remember exactly whether this was really necessary.

https://github.com/eh2k/squares-and-circles/blob/main/lib/rings/dsp/part.h#L64

In general, I think polythony = 1 is actually nice also - this also applies to the following issue: https://github.com/eh2k/squares-and-circles/issues/123.

i'll try it out when i get the chance - maybe i'll manage to introduce a new polythony parameter there.

eh2k, 2025-03-17:
Hi @koswir,

I took a closer look at the rings polyphony setting in vcv. In short, pressing the button switches polyphony to 1, 2, 4.

In squares-and-circles polythony is fixed to 3.
I did this back then to save memory and cpu time. But I can no longer remember exactly whether this was really necessary.

https://github.com/eh2k/squares-and-circles/blob/main/lib/rings/dsp/part.h#L64

In general, I think polythony = 1 is actually nice also - this also applies to the following issue: https://github.com/eh2k/squares-and-circles/issues/123.

i'll try it out when i get the chance - maybe i'll manage to introduce a new polythony parameter there.

koswir, 2025-03-17:
Hey @eh2k

Of course, polyphony = 1 is nice too! But I actually have an original Rings module specifically because I want poly 4, and I’d be happy to let it go (freeing up some HP in my case) if Squares and Circles had Rings poly support.

Also, it doesn’t necessarily have to be a parameter - maybe just separate engine versions like Resonator poly1, Resonator poly2, Resonator poly4? That way, it could be optimized per version without the need for runtime switching.

eh2k, 2025-03-26:
Hi @koswir,

i have pushed a new version yesterday [b1b7d17](https://github.com/eh2k/squares-and-circles/commit/b1b7d17f9950fdd30a5ab2f280c51bae491cf3f8).

You can set the polyphony by the "model" parameter. Now there are 15 Models:

`polythony = 1`

     * Modal - mono
     * Sympath - mono
     * String - mono
     * FM - mono
     * StrQuant. - mono

`polythony = 2`

     * Modal - duo
     * Sympath - duo
     * String - duo
     * FM - duo
     * StrQuant. - duo

`polythony = 4`

     * Modal - quadro
     * Sympath - quadro
     * String - quadro
     * FM - quadro
     * StrQuant. - quadro

theoretically, we now have the same behaviour as with the original rings... (no polyphony = 3 anymore)
The Resonator engine needs a certain amount of memory...3x Resonator should still work. With four I get out of memory on Teensy4.0.

koswir, 2025-03-26:
@eh2k 
maaaan this is sick AF 🤟 
thanks!

closed 2025-10-24 by eh2k

koswir, 2025-11-06:
> [@eh2k](https://github.com/eh2k) [@koswir](https://github.com/koswir) is the chord mode replicated here? I can't figure it out if it is. Big thanks if you're able to help!

Hey. If you mean the alternate mode where you pick a chord with a knob I don't think so. There is a chord mode in M-OSC tho.