eh2k/squares-and-circles #126: Feature Request: Bipolar Attenuator for CV Modulation

koswir, 2025-03-23:
Hey @eh2k 

I've been having fun with the CV modulators recently.
It would be super cool to have a bipolar attenuator knob for AD, ADSR and LFO, though.
It would allow internal amplitude modulation, attenuation or inversion without needing additional modules.

I understand that the o_c's CV outputs are asymmetric but still an inversion could be useful for me.

Thank you for considering this suggestion!
I hope you are doing well!

koswir, 2025-03-31:
@eh2k is it possible to implement it?

eh2k, 2025-04-01:
@koswir - sorry have little time at the moment.

CV engines generally do not have a level parameter in the IO page. Do you think an ‘Attenuator’ (cv-engines) would be logical here instead of Level (audio-engines)?

CV/LFO actually has the ‘Scale’ parameter alleready, which is basically an Attenuator.

koswir, 2025-04-01:
@eh2k 

> [@koswir](https://github.com/koswir) - sorry have little time at the moment.

No! I'm sorry that I tagged you again. I just thought that maybe this topic went unnoticed.

> CV engines generally do not have a level parameter in the IO page. Do you think an ‘Attenuator’ (cv-engines) would be logical here instead of Level (audio-engines)?

No - the parameters in the IO page are not modulable and one has to menu dive there.

> CV/LFO actually has the ‘Scale’ parameter alleready, which is basically an Attenuator.

Yes! exactly!
That is what I need for the envelopes.
The modulable and **bipolar** 'Scale' parameter.
😺

eh2k, 2025-04-02:
Hi @koswir, actually is only about an extra scale parameter for the engines CV/EnvGen_AD and CV/EnvGen_ADSR and presumably also CV/EnvFollower? 

Shouldn't be a big deal...

koswir, 2025-04-03:
> Hi [@koswir](https://github.com/koswir), actually is only about an extra scale parameter for the engines CV/EnvGen_AD and CV/EnvGen_ADSR and presumably also CV/EnvFollower?
> 
> Shouldn't be a big deal...

Yes, thats right!
By bipolar I mean it can invert scaling, so its more like -5/5 range.

eh2k, 2025-04-17:
Hi @koswir - 

just pushed 7ba1769, there are now "AttVer (attenuverter), Offset" parameters in the CV engines:

[image] [image] [image]

The "Vactrol" Mode from the CV/EnvFollowr engine is now a separate eninge:

[image]

koswir, 2025-04-18:
@eh2k Thank you so much!

I know it's not a super exciting thing for you to implement such trivial features, but this is massive!
Super usefull! I love the way you did it!
And both the AttVer and Offset parameters are avaiable for modulation now!
🤟

closed 2025-10-24 by eh2k