eh2k/squares-and-circles #127: Suggestion to add Pan control on the Insert Engines

rootapprox, 2025-03-23:
Hello @eh2k  Hope you doing very well there ! I would love to suggest to add this feature where u can control the Pan of the sounds inserted to an FX , since you cant control the PAN ( or the PAN of the engine wont affect it since you are only feeding the sound to another FX engine without having its output used, I hope i made myself clear here to you , maybe a section for INSERT PAN - (Device) for example after the Insert (Device) Fader for the Vol ? ( the idea is the hove a panned drums inside the inserted FX ) also would be even nicer if we can modulate the PAN in any way .

Thanks a lot !

eh2k, 2025-04-01:
@rootapprox 

thanks for the suggestion.
technically it is feasible, I have to think about the best way to integrate it into the UI.

I'm currently wondering if the pan makes sense in general, since most engines only have mono output and I'm not sure if the reverbs internally sum the inputs as mono anyway.

I'll have to have a look at this in detail when I get the chance

rootapprox, 2025-04-01:
@eh2k 
Thanks a lot !

koswir, 2025-04-02:
> I'm currently wondering if the pan makes sense in general, since most engines only have mono output and I'm not sure if the reverbs internally sum the inputs as mono anyway.

of course pan makes sense 💯some engines are stereo including some of my fxs too 💁