eh2k/squares-and-circles #130: RC-808

eh2k, 2025-04-11:
factory drumkits:
 - [808VinKitOpnCls.zip](https://github.com/user-attachments/files/19710262/808VinKitOpnCls.zip)

eh2k, 2025-04-11:
factory drumkits:
 - [808VinKitOpnCls.zip](https://github.com/user-attachments/files/19710262/808VinKitOpnCls.zip)

eh2k, 2025-04-11:
[CR78.zip](https://github.com/user-attachments/files/19712491/CR78.zip)

eh2k, 2025-04-12:
@rootapprox - thanks for your interest, i wrote something in the first post, i think that answers your question

rootapprox, 2025-04-12:
Thanks @eh2k for having the time to look into this ! 
can we use these packs as multi drums machine ?

eh2k, 2025-04-13:
Now the  □● upload/flash should work. 
Yes, these engines can do ‘multi drums’. However, depending on the drumkit, you may reach the cpu-limits. 

The RC-808 drum kits are not ‘sample-based’. This means that each instrument consists up to 8 layers (partials). 
Each layer is basically an oscillator + filter + several envelopes (see the RC-808 documentation).

The drumsynth-engine is basically a recreation (written c++ code by me - work-in-progress) of the RC-808 concept / building blocks.
  - It's a lucky coincidence that the RC-8080 drumkits are simple xml files that are relatively easy to read.
  - screenshot from the rc-808 reference manual:
  - [image]

Acknowledgments & Credits:
 - http://rc808.xsrv.jp/wp/about-us/

koswir, 2025-04-18:
WTF just happend here! 😄 how can I work now?!!!!

koswir, 2025-04-19:
@eh2k 

> depending on the drumkit, you may reach the cpu-limits.

are there any guidelines how to reduce cpu load?
are unused partials disabled?
what about disabling elements of the signal patch like setting filers or wavetable to "thru"?

eh2k, 2025-04-20:
@koswir 

In general, it depends on the partial count. Blocks that are set to THRU do not cost any CPU time. With wave-shaping blocks, the more points on the curve so probably more CPU calculation. The VCF block, for example, is not yet implemented (it was not used anywhere in the factory drumkits). 

The multi-trigger mode is polyphonic, i.e. if several triggers occur simultaneously, all instruments must be rendered in the same time frame. If an instrument is not triggered or the envelopes run out, no CPU is used.

rootapprox, 2025-04-20:
@eh2k Hey i noticed that when i run 2 DrumsSynth engines ( using them with multi triggers ) a big noise and like heavy bit reductions happens , so i guess its cause the Cpu limit right ?

eh2k, 2025-04-20:
@rootapprox 

yes - sounds like CPU limit.

The 909 drumkit in particular is CPU-intensive. The toms and hihats have 6 to 8 partials. If the CPU is at its limit, you should hear a kind of bit reduction / time-stretch with all engines (e.g. resonator is also cpu-heavy).

koswir, 2025-04-20:
> [@koswir](https://github.com/koswir)
>
> In general, it depends on the partial count. Blocks that are set to THRU do not cost any CPU time. With wave-shaping blocks, the more points on the curve so [...]

@eh2k Cool!
great tips! so a low cpu kit should use very few partials :)
would be super nice to implement the vcf block because it can be modulated while biquads seem static

one more thing:
whats wrong with longer sounds like cymbals?
they seem cutted

eh2k, 2025-04-22:
@koswir 

thanks for the hint - i hadn't noticed it until now - drums are not rendered after 1.3s... ([DrumSynth.cpp#L103](https://github.com/eh2k/squares-and-circles/blob/7ba1769b503948d06e0a4dc6fb96da78eaf6b888/app/DRUMS/DrumSynth.cpp#L103)) - this will be fixed with the next update.

Here are a few more implementation details:

Originally a variant of drumsynth was introduced for the claps engine. There are a list of envelopes and filter settings that is randomly shuffled. The seeds for interesting sounding claps are hardcoded.

The Color parameter affects the oscillator frequency and also the frequency of the biquads (because clap is primarily generated from noise, which is not pitchable) - Modulating color may be cpu-consuming here..

The Decay parameter, stretches all envelopes.

The stereo parameter (actually only for noise oscillators) is basically the mix of the left and right noise oscillator.

Two further parameters would be conceivable. I was thinking of something for the wave-shaper.

koswir, 2025-04-23:
> Here are a few more implementation details:
> Originally a variant of drumsynth was introduced for the claps engine. There are a list of envelopes and filter settings that is randomly shuffled. The seeds for interesting sounding claps are hardcoded.

Are these randomly shuffled setting the part of the new engine with sound import? 
or is that just a rant about the clap engine?
😸 

> The Color parameter affects the oscillator frequency and also the frequency of the biquads (because clap is primarily generated from noise, which is not pitchable) - Modulating color may be cpu-consuming here..

Do you mean the Pitch parameter?

> The Decay parameter, stretches all envelopes.
> The stereo parameter (actually only for noise oscillators) is basically the mix of the left and right noise oscillator.
> Two further parameters would be conceivable. I was thinking of something for the wave-shaper.

I would vote for:

Pitch (oscs pitch + biquads)
Sound select
Filter (vcf cutoff)
Stereo
Decay (all envelopes except pitch and filter)
Mod Decay (pitch and filter env scaling)

koswir, 2025-04-24:
[DnB lowcpu.zip](https://github.com/user-attachments/files/19900242/DnB.lowcpu.zip)
I made a low cpu kit - mostly 1 partial per drum except snares.
Can't wait to try it in □︎●︎
🐱

koswir, 2025-04-24:
@eh2k 

> [DnB lowcpu.zip](https://github.com/user-attachments/files/19900242/DnB.lowcpu.zip) I made a low cpu kit - mostly 1 partial per drum except snares. Can't wait to try it in □︎●︎ 🐱

I just tested my kit in the drumsynth kit uploader. All parts with the metalic osc (Meta Hc, Ride 1, Ride 2) didn't render - I can not preview them.
Also sounds are a little different than in the rc808 (especially snares) 🤔

koswir, 2025-04-26:
@eh2k 
> The sounding difference
> 
> * this is probably due to the waveshaper, there is probably something wrong with more complex curves...
> * or the missing vcf implementation..

yep, you are right.
Both cases actually.
I managed to get closer to the kit I created.

eh2k, 2025-05-19:
@koswir,

I have optimised/fixed the waveshaper in cb81c7c. The curve should now be pretty close to the RC808. I have introduced a debug mode for the DrumSynth web app. It is possible to modify the parameters also - unfortunately flashing is not yet possible in debug mode (it is experimental - I don't know if I want to develop in that direction) .

https://eh2k.github.io/%E2%96%A1%E2%97%8F/drumsynth/?debug

I have noticed that the audio playback in Crome is somehow filtered and sounds different. Therefore there is now a possibility to download the drums as an audio file, this should be 100% exactly like on the S&C.

koswir, 2025-06-15:
@eh2k 
Hey man I hope you are doing great.
Is there any chance of adding V/Oct tracking to the DrumSynth?
👋

koswir, 2025-07-08:
s&c machines are mostly mono - but you can open 4 of them and map midi channels

koswir, 2025-07-08:
btw
@eh2k any chance for v/oct implementation for the drumsynth? so chromatic playing is possible (eg tuned kick/bass)

eh2k, 2025-07-08:
@koswir 

Yes, it shouldn't be a big deal. Unfortunately other  priorities at the moment, you know summer, sun, vacation and most of the time away from eurorack & laptop...

eh2k, 2025-07-08:
@MidiSlave 
 - in the IO-Page you can assign "MIDI-Channel" to an engine  (you have also to set the MIDI-Control to MIDI/USB in die CLK-Midi-Settings).
 - the key/drum mapping: https://github.com/eh2k/squares-and-circles/blob/main/app/DRUMS/DrumSynth.cpp#L133 (not complete at the moment)

MidiSlave, 2025-07-08:
I got it working. Had to set all voices to a trig / cv input in order for
them to be triggered via midi.

You're a legend :)

Here's some things that I discovered / changed along the way:

### 1. **MIDI Triggers Never Cleared**
**Problem**: Once a MIDI note triggered a drum voice, the `_midi_trigs`
bitmask was never cleared, causing infinite retriggering.

**Location**: `app/DRUMS/DrumSynth.cpp:99-104`

```cpp
// BEFORE:
if (engine::trig() & (1 << i) || _midi_trigs & (1 << i))
{
_t[i] = 0;
drum_synth_reset(_inst[i]);
}

// AFTER:
if (engine::trig() & (1 << i) || _midi_trigs & (1 << i))
{
_t[i] = 0;
drum_synth_reset(_inst[i]);
// Set velocity for trigger inputs (full velocity)
if (engine::trig() & (1 << i)) {
_midi_velocity[i] = 1.0f;
}
_midi_trigs &= ~(1 << i); // Clear MIDI trigger after processing
}
```

#### Added velocity storage:
```cpp
// Line 45: Added velocity array
float _midi_velocity[16] = {}; // Store velocity for each voice
```

#### Updated all MIDI note cases to store velocity:
```cpp
// Example for each case:
case 35: // BD0
_midi_trigs |= (1 << 0);
_midi_velocity[0] = velocity / 127.0f; // Store normalized velocity
break;
```

#### Applied velocity scaling in audio processing:
```cpp
// Lines 115-122: Apply velocity to mix levels
// Apply velocity scaling for MIDI-triggered notes
float velocity_scale = 1.0f;
if (_midi_velocity[i] > 0.0f) {
velocity_scale = _midi_velocity[i];
}

params.levelL = engine::mixLevelL(i) * velocity_scale;
params.levelR = engine::mixLevelR(i) * velocity_scale;
```

#### Reset velocity when voices finish:
```cpp
// Lines 134-137: Clean up velocity when voice stops
else
{
// Voice finished playing - reset velocity
_midi_velocity[i] = 0.0f;
}

On Wed, Jul 9, 2025 at 12:06 AM eh2k ***@***.***> wrote:

> *eh2k* left a comment (eh2k/squares-and-circles#130)
> <https://github.com/eh2k/squares-and-circles/issues/130#issuecomment-3049113374>
>
> @MidiSlave <https://github.com/MidiSlave>
>
>    - in the IO-Page you can assign "MIDI-Channel" to an engine (you have
>    also to set the MIDI-Control to MIDI/USB in die CLK-Midi-Settings).
>    - the key/drum mapping:
>    https://github.com/eh2k/squares-and-circles/blob/main/app/DRUMS/DrumSynth.cpp#L133
>    (not complete at the moment)
>
> —
> Reply to this email directly, view it on GitHub
> <https://github.com/eh2k/squares-and-circles/issues/130#issuecomment-3049113374>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/AZDTMDXUL2F5NMBLODBZHJD3HPF6BAVCNFSM6AAAAAB26TTPQCVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZTANBZGEYTGMZXGQ>
> .
> You are receiving this because you were mentioned.Message ID:
> ***@***.***>
>

eh2k, 2025-07-08:
@MidiSlave 

Thank you for figuring that out.

I can't say exactly how I tested this. The `_midi_trigs` clear/reset really seems to be missing.
It should be more or less the same logic as for TR707 here: https://github.com/eh2k/squares-and-circles/blob/main/app/DRUMS/TR707.cpp#L95

Probably because I wanted to optimize the performance of the drum synth and somehow got confused.

By the way, here is the "cv to pitch" line: https://github.com/eh2k/squares-and-circles/blob/main/app/DRUMS/DrumSynth.cpp#L109

koswir, 2025-07-08:
> [@koswir](https://github.com/koswir)
> 
> Yes, it shouldn't be a big deal. Unfortunately other priorities at the moment, you know summer, sun, vacation and most of the time away from eurorack & laptop...

Of course! :D I am going to take my modular to the seaside tho!

eh2k, 2025-10-24:
- Midi should work now...in latest c745be4
  - V/OCT in case of "single instrument mode" 
  - MultiTrigger by Keys 60-76 or GM Standard Drum (maybe confusing)
- Also the "waveshaper crash #140" is fixed 
 - The waveshaper has been reimplemented - it should now be very close to the original sound.  

Regarding MIDI implementation, I have added this to the readme file:

----

* **Midi-Channel**: `-`, `1`...`16`
  - The Midi messages are translated to analog events
    - Trig: NoteOn
    - Gate: NoteOn until NoteOff
    - CV: 0V = Key 60
    - Velocity is not supported..
  - [Multi-Trig-Drum-Engines](https://github.com/eh2k/squares-and-circles/wiki/DrumSynth)
    - In case trigger to `-` - individual drums/instruments are triggered by individual keys:
        - GM Standard Drum Map: https://github.com/eh2k/squares-and-circles/wiki/TR707#midi-mapping
        - or Keys 60-75 for instrument 1-16
        - Velocity is not yet supported..