eh2k/squares-and-circles #131: -60 db noise floor

rootapprox, 2025-04-24:
Hello @eh2k  can i ask you please if its okay to have that amount of noise floor coming from the module ? is calibrating the module would help with that ? ( since i never calibrated the module before even when i installed this firmware ) thanks for any hints

koswir, 2025-04-24:
@rootapprox 
The noise floor on a dac of your o_c may be caused by a specification of your system or how your o_c was built. One can use different parts to build o_c - for example low noise preamps before a dac.

My buddy who did build all my o_cs told me that if your preamps are tlO72 - it may be noisy. You can check that on your o_c. I got low noise opa2277 but opa2172 are also good.

The dacs are 16bit Dac8565icpw so theoretical headroom should be about 90db.

The calibration process is to calibrate input and output voltages eg that 1v = 1v
so it is not going to help you with the signal quality.

The inputs are expecially noisy which is a known issue.

rootapprox, 2025-04-24:
@koswir Thanks a lot for the informations and the hints about why im experiencing this , 
im using this Module : https://www.tlm-audio.com/product-page/oc-t4-0-tlm-audio-eurorack-module-ornament-crime-teensy-4-0
I just sent them a mail to see whats the DAC type that i have the ...64 or ...65 , as i saw some options to change that in the firmware webflasher, i thought maybe choosing the right type of DAC would help with the noise ..

I appreciate any other hints to eliminate this noise and have the best experience with this firmware .. <3

rootapprox, 2025-04-24:
@eh2k Thanks for the infos ! yes im goin to be waiting their response and if that changes anything when changin the DAC 64 to 65 .. i saw that you had the modules you build for sale on modular grid can u confirm that ? i would simply bought it from you since i just used the other firmwares for about hours before changing to your amazing work ! 

thanks a lot!

rootapprox, 2025-04-24:
@koswir after opening the back of the module i saw that i have this opa2172 in the module .. so i think the problem maybe not in that DAC ? 

[image]

koswir, 2025-04-25:
@rootapprox 
These MCP60021 are the middle ones.
Please check out the ones at the bottom of your device.

rootapprox, 2025-04-25:
the ones in the bottom of my device are this little chip :  02172A TI 43K AH34

koswir, 2025-04-25:
> the ones in the bottom of my device are this little chip : 02172A TI 43K AH34 with DAC 8565 3AEE in the top chip

This is the other one my builder confirmed as high quality - 2172
Maybe its your power supply or other modules 🤔

koswir, 2025-04-25:
@rootapprox 
My buddy said that you can check his builds on reverb. He is haveing a week sale right now.
If you are in europe you can but one to compare with your build then return it in 14 days.
https://reverb.com/en-pl/shop/helenas-gear-garage-21

rootapprox, 2025-04-25:
@koswir thanks a lot for the follow up ! Im using [Sinusoda](https://sinusoda.com/) case which is known for low noise from what i hear .. https://cdn.modulargrid.net/img/racks/modulargrid_2688585.jpg  this is my rack entirely.. i do keep it in the middle fader on the mixer engines for amplyfing it .. yes its not really audible at jamming but when i master the tracks i can hear some crackles which is sounding like a vinyl crackles sound .. also i noticed that the more modulation externally i do in the engines the more that movement of crackles happens .. 

about the reverb link , do you confirm that this module wont have that crackles and noise ?