eh2k/squares-and-circles #132: noo MIDI since firmware update

kaosbeat, 2025-04-25:
Hi, 

I flashed my OnC snc with the latest firmware. I was occasionally expoeriencing a midi device not found, but unplugging replugging consistently fixed that for me. I was only when the device was plugged in on boot it sometimes failed. I suspected a power issue  ( both the Rapsberry Pi  the OnC is connected to and the modular are powered on by the same switch.

I read something in the changelog on 5 second tiomeout for midi, and I hoped that was a fix for this problem.

I used the webflasher on my laptop and tried upgrading to latest
This evenetually succeeded but in about a third of the progress the progress bar started moving really slow. Only a handfull machnies were installed
I reverted to the 20250125 firmware then, (Lowest I could find in webflasher) Flashing was much faster (10X) but now the device is no longer recognised as a MIDI device. I did check the settings.

dmesg says (the first is the succesful coinnect from before the flashing):

```

5986.554677] usb 3-8: USB disconnect, device number 14
[85986.894129] usb 3-8: new high-speed USB device number 15 using xhci_hcd
[85987.103722] usb 3-8: New USB device found, idVendor=16c0, idProduct=0489, bcdDevice= 2.79
[85987.103725] usb 3-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[85987.103726] usb 3-8: Product: S&C 04E9E51758E3
[85987.103728] usb 3-8: Manufacturer: eh2k
[85987.103729] usb 3-8: SerialNumber: 15300830
[85987.145760] cdc_acm 3-8:1.0: ttyACM0: USB ACM device
[85987.151723] usb 3-8: Quirk or no altset; falling back to MIDI 1.0
[86001.041846] usb 3-8: USB disconnect, device number 15
[86182.819031] usb 3-8: new high-speed USB device number 16 using xhci_hcd
[86188.315922] usb 3-8: device descriptor read/64, error -110
[86204.187505] usb 3-8: device descriptor read/64, error -110
[86204.473457] usb 3-8: new high-speed USB device number 17 using xhci_hcd
[86209.819355] usb 3-8: device descriptor read/64, error -110
[86225.690939] usb 3-8: device descriptor read/64, error -110
[86225.802869] usb usb3-port8: attempt power cycle
[86226.240887] usb 3-8: new high-speed USB device number 18 using xhci_hcd
[86231.040997] usb 3-8: Device not responding to setup address.
[86236.054016] usb 3-8: Device not responding to setup address.
[86236.261624] usb 3-8: device not accepting address 18, error -71
[86236.439621] usb 3-8: new high-speed USB device number 19 using xhci_hcd
[86241.239018] usb 3-8: Device not responding to setup address.
[86246.245043] usb 3-8: Device not responding to setup address.
[86246.453368] usb 3-8: device not accepting address 19, error -71
[86246.458307] usb usb3-port8: unable to enumerate USB device
[86256.427100] usb 3-8: new high-speed USB device number 20 using xhci_hcd
[86262.042990] usb 3-8: device descriptor read/64, error -110
[86277.914594] usb 3-8: device descriptor read/64, error -110
[86278.198534] usb 3-8: new high-speed USB device number 21 using xhci_hcd
[86294.292114] usb 3-8: new high-speed USB device number 22 using xhci_hcd
[86299.931007] usb 3-8: device descriptor read/64, error -110
[86315.800593] usb 3-8: device descriptor read/64, error -110
[86316.086547] usb 3-8: new high-speed USB device number 23 using xhci_hcd
```

stuck now, wwhat should I do?
I have a screen shot from during the slow/failing flashing

[image]

I also tried loading one of my saved patches, this failed. Should this work?

kaosbeat, 2025-04-25:
Ok, I swapped the cable, tried power cycling PC. All very ionconsisent results.
Eventually I succeeded in reflashing to the latest, which now seems to have all(?) machines, but still no midi connection.

I also tried connecting the teensy separated from the OnC but it does not even power on. Is this expected behaviour?

I'm gonna try and flash the hex with teensy loader now

kaosbeat, 2025-04-25:
ok, the hex file apparently is only for the loader, so I did not try that.
I was again able to interact with the device again after I pressed te reset button and powercycles a few times,
 flashed it using the webinterface to the 20250111 firmware with the link from the releases page.

Nowe it identifies as 
```

  265.299982] usb 1-1.4: USB disconnect, device number 10
[  268.357254] usb 1-1.4: new high-speed USB device number 11 using xhci-hcd
[  268.470891] usb 1-1.4: New USB device found, idVendor=16c0, idProduct=0478, bcdDevice= 1.08
[  268.470918] usb 1-1.4: New USB device strings: Mfr=0, Product=0, SerialNumber=1
[  268.470926] usb 1-1.4: SerialNumber: 001758E3
[  268.532591] hid-generic 0003:16C0:0478.0004: hidraw1: USB HID v1.11 Device [HID 16c0:0478] on usb-xhci-hcd.3.auto-1.4/input0

```

but stil no (consistent) midi.

I think I ruled out the cable.
The connector inside the modular is not moving when I connect/disconnect the cable, it passes through a point where the cable is fixed...

I tried multiple computers/cables/...

I used to have consistency with the hardware, I did not touch any of the hardware until after the problem started. Could this be software related, or just bad luck?

kaosbeat, 2025-04-25:
After more testing, I have found consistent behavior. 

After a power cycle the device is correctly detected.
After an usb unplug /replug this is no longer the case

This is consistent.

So to reproduce the (not) detection.
1. Power on the O_c while plugged into PC1 that is powered on at the same time. No device detected
2. with the O_c still powered on, plug in into PC2. No Device detected
3. with PC2 still powered on, power cycle the O_c Device detected
4. plug it back into PC1, no device detected
5.  power cycle O_c, device detected on PC1

So it's not the cable.

Is there a way to reboot the device without power cycling? My PC,, as I mentioned is built into the case and has tha same power cycle button

koswir, 2025-04-25:
I have also added a new post how to nicely connect s&q o_cs to midi over T1 jack.
https://github.com/eh2k/squares-and-circles/issues/133

kaosbeat, 2025-04-25:
weird thing is, first time after power cycle it always works. Second time no more... 
I have no option for TRS midi. I'm connecting to a miniPC having only USB, no TRS/DIN midi. but your solution looks nice though :-)

koswir, 2025-04-25:
> weird thing is, first time after power cycle it always works. Second time no more...

yep its weird - lets wait for @eh2k 

> I have no option for TRS midi. I'm connecting to a miniPC having only USB, no TRS/DIN midi. but your solution looks nice though :-)

well - an audio/midi interface or a midi interface connected to it's usb port should do the trick 🐈

kaosbeat, 2025-04-25:
Ok, sounds reasonable as an explanation. I will try the reset procedure and see if that helps...

pssssssst, 2025-05-13:
I also have problems with the USB-Midi feature. It works all the time on my PC but just barely on my groovebox. When it works, it's great though.
The other teensy 4.0 o_c firmware, phazerville, doesn't have these connectivity issues. It's not that important to me but it would be cool if this was reliably working.

kaosbeat, 2025-05-13:
I can confirm that when I boot my machine, powering on both O_c and Linux SBC at the same time, that when I reset the O_c when the Linux box is almost booted, I have zero problems. Never.

So the problem occurs, I think, when booting the O_c when the host is not ready yet.

pssssssst, 2025-05-13:
I have to boot the o_c first, before even turning on my groovebox, because the groovebox scans for client devices only on startup. I'll have to investigate further.

kaosbeat, 2025-05-13:
> I have to boot the o_c first, before even turning on my groovebox, because the groovebox scans for client devices only on startup. I'll have to investigate further.

Try turning on the O_c and right before you turn on the groove box reset it by pressing the two right buttons ( right encoder and right button).

If you get the timing right it could work...

pssssssst, 2025-05-13:
Hmm, that's definitly worth a try. I will give it a shot tommorow

eh2k, 2025-05-14:
I think I've figured it out now ... but I have no idea what exactly is going wrong.

In any case, you can reproduce the issue as follows:
 - Start the device with right-button pressed (patch restore skipped - no engines loaded).
    - usb is working fine:  `lsusb -v -d 16c0:0489` -> `Device Status:     0x0000`
    - disconnect/connect usb cable -> usb ist working, `lsusb` is listing etc..
- If any engine loaded
   -   `lsusb -v -d 16c0:0489`  -> `cannot read device status, Resource temporarily unavailable (11)`
   - disconnect/connect usb cable:
         - `dmesg` -> `usb 1-2: device descriptor read/64, error -110.`
         - `lsusb` no response...

The cause is writing the engine ram_base to `0x20000000`  (details see https://github.com/eh2k/udynlink/blob/master/README.md)

pssssssst, 2025-05-15:
In a

> Try turning on the O_c and right before you turn on the groove box reset it by pressing the two right buttons ( right encoder and right button).
>
> If you get the timing right it could work...

That does the trick. If i turn on my groovebox while the o_c is on it's boot screen it works everytime.
Thank you :)

kaosbeat, 2025-05-20:
confirmed fixed! 🙏🏼

closed 2025-05-20 by kaosbeat