eh2k/squares-and-circles #134: Default Parameter CC-Mappings question

koswir, 2025-04-27:
hey @eh2k 
Default Parameter CC-Mappings
| HEX | DEC | parameter-index | CH |
|-----|-----|-----------------|----|
|  20 |  32 |        0        |  * |
|  21 |  33 |        1        |  * |
|  22 |  34 |        2        |  * |
|  23 |  35 |        3        |  * |
|  24 |  36 |        4        |  * |
|  25 |  37 |        5        |  * |
|  26 |  38 |        6        |  * |
|  27 |  39 |        7        |  * |

If all included engines have max 6 parameters.
What are exactly 6 and 7?

If they are there just in case, would it be possible redirect them to pan and level?

eh2k, 2025-05-04:
hi @koswir,

I checked the source code and have to say that there are still some things missing from the readme regarding midi.

I will add it to the readme next update.
In short, there is a default midi handler where note messages are translated to the triggers, gate, accent and CV and the CCs are translated to the parameters. (the midi handling can also be implemented individually in the engine itself)

#### Default Midi Handler:
- Program Change -> Engine select
- Trigger -> NoteOn,
- Gate -> NoteOn until NoteOff
- Accent ->  triggered when velocity > 100. (hihat engines usecase ?!)
- CC 07 -> volume control (io-page)
- CC 32-39 -> for the parameters 0-7 
   - theoretically there can be up to 24 parameters internally, but this is untested, ui would then have to display a list like in the io-settings instead of the encoders

There is a table regarding CC numbers and the functions behind them that I would like to follow: https://midi.org/midi-1-0-control-change-messages.

Regarding Pan:
 - I would map/implement the pan to CC 10 next firmware update.
 
Further suggestions welcome, there has already been a request to control modulations via Midi CC https://github.com/eh2k/squares-and-circles/issues/11.

eh2k, 2025-05-19:
- [x] PAN CC implemented in [cb81c7c](https://github.com/eh2k/squares-and-circles/commit/cb81c7cc310620d4d463212b81f5fa0303ca1eb1)
- [x] Readme.md updated -> https://github.com/eh2k/squares-and-circles?tab=readme-ov-file#midi-control

closed 2025-10-24 by eh2k