eh2k/squares-and-circles #136: Chrash report

pssssssst, 2025-05-12:
I recently flashed a firmware update because USB-Midi stopped working all of a sudden.
After the update i get this crash report.
I can still access the Teensy via Web-Midi, but nothing else.
Any ideas?
Thanks for your work, i was haviong alot of fun with this build.

[image]

pssssssst, 2025-05-12:
Nevermind. I reset the teensy by pushing the button for exactly 15 seconds, now it works again including USB-MIDI

eh2k, 2025-05-12:
Thanks for posting your solution. 

Then there was probably something in the EEPROM memory area that caused the crash. The long press should reset the entire flash memory, as I remember. 

Possible holding the right button at startup would have helped also (skip last saved patch restore, and start from scratch)

closed 2025-05-12 by eh2k